MicrosoftDocs/windows-driver-docs-ddi #1521: Customer feedback - Incomplete documentation

sandilya-xilinx, 2024-01-18:
### Type of issue

Missing information

### Feedback

Can you please elaborate about the members of D3DDDI_SEGMENTPREFERENCE

### Page URL

https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/d3dukmdt/ns-d3dukmdt-_d3dddi_segmentpreference

### Content source URL

https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/d3dukmdt/ns-d3dukmdt-_d3dddi_segmentpreference.md

### Author

@lorihollasch

lorihollasch, 2024-01-23:
Thanks for reporting this blank page, @sandilya-xilinx - oops! I've populated the page with details. :)

closed 2024-01-23 by lorihollasch

sandilya-xilinx, 2024-01-23:
Thank you @lorihollasch for updating the page. Also, I want to thank MSFT to provide feedback link which is easy to submit our request/feedback for the MSDN pages and the turnaround time is impressive. Thank you again.