MicrosoftDocs/windows-driver-docs-ddi #1533: Customer feedback -

kouzhudong, 2024-01-29:
### Type of issue

Typo

### Feedback

PVOID                               CallbackRegistration
should be
PVOID *                             CallbackRegistration

### Page URL

https://learn.microsoft.com/zh-CN/windows-hardware/drivers/ddi/wdm/ns-wdm-_io_session_state_notification

### Content source URL

https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/ns-wdm-_io_session_state_notification.md

### Author

@barrygolden

tedhudek, 2024-02-02:
Is this reporting the same issue as #1534?  I don't see CallbackRegistration on https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/ns-wdm-_io_session_state_notification.  Please re-open if I am missing something.

closed 2024-02-02 by tedhudek

kouzhudong, 2024-02-03:
here is true issue.