MicrosoftDocs/windows-driver-docs-ddi #1534: Customer feedback -

kouzhudong, 2024-01-29:
### Type of issue

Typo

### Feedback

PVOID CallbackRegistration
should be
PVOID * CallbackRegistration

### Page URL

https://learn.microsoft.com/zh-CN/windows-hardware/drivers/ddi/wdm/nf-wdm-ioregistercontainernotification

### Content source URL

https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-ioregistercontainernotification.md

### Author

@barrygolden

tedhudek, 2024-02-02:
Hi @kouzhudong, thanks for the report.  I do not repro that in the 22621 wdm.h, though.  Can you tell me more about how you found this?

```
NTSTATUS
IoRegisterContainerNotification(
    _In_ IO_CONTAINER_NOTIFICATION_CLASS NotificationClass,
    _In_ PIO_CONTAINER_NOTIFICATION_FUNCTION CallbackFunction,
    _In_reads_bytes_opt_(NotificationInformationLength) PVOID NotificationInformation,
    _In_ ULONG NotificationInformationLength,
    _Out_ PVOID CallbackRegistration
    );
```

kouzhudong, 2024-02-03:
https://learn.microsoft.com/zh-CN/windows-hardware/drivers/ddi/wdm/nf-wdm-ioregistercontainernotification

here is true issue.

tedhudek, 2024-02-07:
On both these pages, https://learn.microsoft.com/zh-CN/windows-hardware/drivers/ddi/wdm/nf-wdm-ioregistercontainernotification (zh-ch version) and https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-ioregistercontainernotification (en-us version), it shows ` [out]          PVOID                               CallbackRegistration`, which matches the header (see above).  What is the specific problem?

kouzhudong, 2024-02-07:
CallbackRegistration not PVOID 
CallbackRegistration is PVOID *

tedhudek, 2024-02-12:
Fixed!  Thanks for the feedback.

closed 2024-02-12 by tedhudek