MicrosoftDocs/windows-driver-docs-ddi #1537: Please document cache aware rundown protection functions

ori-l, 2024-02-05:
### Type of issue

Missing information

### Feedback

Hello!
Cache aware rundown protection functions is crucial to increase performance for drivers. It allows for a rundown protection to be aware of cache overrides caused by multi-processor computers.

When looking at the current official documentation:

The structure `EX_RUNDOWN_REF_CACHE_AWARE ` is defined in "wdm.h" on 10.0.18362.0 SDK:
```
//
//  Opaque cache-aware rundown ref structure
//

typedef struct _EX_RUNDOWN_REF_CACHE_AWARE  *PEX_RUNDOWN_REF_CACHE_AWARE;
```

The exact definition of the structure is not defined an not documented anywhere which can cause misuse and crashes in worst case scenarios. 

The functions that use the structure - `ExInitializeRundownProtectionCacheAware`, `ExAcquireRundownProtectionCacheAware\Ex`, `ExReleaseRundownProtectionCacheAware\Ex`, `ExWaitForRundownProtectionReleaseCacheAware`, and `ExReInitializeRundownProtectionCacheAware` are barely documented and it is unclear at first glance what the difference between them and the normal rundown protection functions, and how to safely use them. It is not possible to invoke any of these functions without defining the structure.

Since these functions and structure already appear on official documentations, can you please add the definition of the structure and more detailed explanation on the functions for safe usage?

### Page URL

n/a

### Content source URL

n/a

### Author

n/a

tedhudek, 2024-02-26:
Thanks for pointing this out, @ori-l

I just published updates to cover cache aware rundown protection.  For an overview, see the following new section -- 
https://learn.microsoft.com/en-us/windows-hardware/drivers/kernel/run-down-protection#cache-aware-run-down-protection

In addition, the following API ref pages now have more info than they did previously:

* https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-exacquirerundownprotectioncacheawarelearn
* https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-exacquirerundownprotectioncacheawareexlearn
* https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-exallocatecacheawarerundownprotectionlearn
* https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-exfreecacheawarerundownprotectionlearn
* https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-exinitializerundownprotectioncacheawarelearn
* https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-exreinitializerundownprotectioncacheawarelearn
* https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-exreleaserundownprotectioncacheawarelearn
* https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-exreleaserundownprotectioncacheawareexlearn
* https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-exrundowncompletedcacheawarelearn
* https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-exsizeofrundownprotectioncacheawarelearn
* https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-exwaitforrundownprotectionreleasecacheawarelearn

closed 2024-02-26 by tedhudek

ori-l, 2024-02-28:
Great, thank you!