MicrosoftDocs/windows-driver-docs-ddi #1539: Customer feedback - Can we have the missing structures please?

rodwiddowson, 2024-02-06:
### Type of issue

Typo

### Feedback

We have recently tripped over this
``` FileStatBasicInformation,                       // 77```

It would have been super handy to have the structure to hand (since it doesn't appear in WDK 10.0.22721 which is the latest...

### Page URL

https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/ne-wdm-_file_information_class

### Content source URL

https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/ne-wdm-_file_information_class.md

### Author

@barrygolden

rodwiddowson, 2024-02-06:
FYI: I found the missing structures in the Canary WDK build preview (10.0.26040.0)

lorihollasch, 2024-02-06:
@rodwiddowson, thanks for flagging this. With thousands and thousands of pages, it's unfortunately easy for us to miss cross-referencing linked material. I wrote the [structure page](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/ntifs/ns-ntifs-file_stat_basic_information) for FileStatBasicInfo last week as prerelease info if that helps unblock. Will start working on the rest ...

lorihollasch, 2024-02-07:
I just published the other missing structs and linked them from FILE_INFORMATION_CLASS!

closed 2024-02-07 by lorihollasch