MicrosoftDocs/windows-driver-docs-ddi #1540: Customer feedback -

xiulipan, 2024-02-07:
### Type of issue

Typo

### Feedback

> [out] pBankRangeTable
> 
> An array of values that indicates the ranges that delimit each bank in the segment. The array specifies the end addresses of the first bank through the nâˆ’1 bank (that is, the end offsets into the segment for each bank). Note the following:

What is the nâˆ’1 means here

### Page URL

https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/d3dkmddi/ns-d3dkmddi-_dxgk_segmentdescriptor

### Content source URL

https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/d3dkmddi/ns-d3dkmddi-_dxgk_segmentdescriptor.md

### Author

@lorihollasch

xiulipan, 2024-02-07:
or what the language is using here

lorihollasch, 2024-02-08:
Thanks for reporting the weird character, @xiulipan. That should be a reference to the NbOfBanks member - I've updated the page with a fix.