MicrosoftDocs/windows-driver-docs-ddi #1543: Issues to be discontinued March 20Feedback

tedhudek, 2024-02-16:
### Type of issue

Other (describe below)

### Feedback

Hello Driver Community,

On Wednesday March 20, we will discontinue the option to provide feedback on the driver docs using GitHub issues. This system is being replaced by a new feedback control that appears as a thumbs up button with the word Feedback in the upper right of each driver documentation page. When you select this icon, it expands to allow you to pick thumbs up or down, as well as from a variety of top reasons why, and then a box where you can include a verbatim. The new system is already live, so you can use it today. Just as with GitHub issues, any feedback that you provide using the new method still makes its way to the writer who owns that page.

You can submit comments in languages other than English, and submissions remain anonymous. This form is consistently available across Microsoft Learn articles and training modules. Please note that pull requests are still accepted for content. To start a pull request, select the pencil icon in the upper right corner of the article. (above the Feedback button)

For more info about this change, refer to this blog post - https://techcommunity.microsoft.com/t5/microsoft-learn-blog/announcing-a-new-way-to-give-feedback-on-microsoft-learn/ba-p/4027635

We appreciate the contributions from the driver development community and look forward to continued collaboration.

### Page URL

a

### Content source URL

a

### Author

a

Neme12, 2024-02-25:
Seriously? I thought MS was being more and more open, and now you're going back. :(

tedhudek, 2024-03-02:
Hi @Neme12, thanks for commenting.  It is indeed a big change.  What I can commit to you is that we do look at each and every verbatim that we receive in the new system.  Some are actionable, some are not.  For the ones where we make updates, it might be a few weeks or even months, but we do track the work for these and address it as we can.  We hope you'll continue to send in feedback!

Neme12, 2024-03-03:
@tedhudek The difference is - on GitHub, people can not only create issues but also fix them.

tedhudek, 2024-03-05:
Hi @Neme12, we actually haven't had any cases I'm aware of where one customer reported an issue and a different customer proposed a fix, if I am understanding your scenario. You could do a similar thing with PRs, which remain supported and even encouraged.